df8oe/UHSDR #1845: Menu item missing

WD8BXS, 2020-04-16:
The touchscreen menu is not listed in the menu over view.
Also I can't even get to the menu overview right now, maybe some one is working on it  

Chuck

df8oe, 2020-04-17:
Hi Chuck,

I cannot confirm. On all of my radios the menu item **is** available in the menu overview and it is working. All of my radios do have touchscreen function since years so I never have taken a look on that. But maybe that is the difference: does your radio support touch function?

vy 73
Andreas

WD8BXS, 2020-04-17:
Hi Andreas,  sorry , I meant on the automatically generated menu overview on the Wiki
[image]

WD8BXS, 2020-04-17:
The link does not work, and the overview does not include the touchscreen menu,

Thanks,
 Chuck

df8oe, 2020-04-17:
That pointed me to the real issue ;)

df8oe, 2020-04-19:
Please close the issue when it is solved for you...

WD8BXS, 2020-04-19:
The link works now, but there is still no touch screen menu listed.

df8oe, 2020-04-20:
Can you remeber: has that ever been included before?

WD8BXS, 2020-04-20:
Sorry, I do not remember, maybe never!!

WD8BXS, 2020-04-21:
I know nothing about coding, but if you look at it, it looks like the touchscreen menu is supposed to be there.

df8oe, 2020-04-21:
Hi Chuck,

I have taken a quick look at the menu_structure and the python script and I do not see any reason why it should not work. I already have checked the placements (chain) - all ok. I have renamed it (maybe the number harms - no, that' not the reason). So I think it is a problem regarding python and regular expressions (which are not my best friends ;) ). So I will leave this due to a lack of time and minor priority. Maybe someone else fetches this...

db4ple, 2020-04-21:
The graphics already included the touch menu, it was only the table which missed the data. Reasonit uses an list of top level menu entries to order content, and MENU_MEN2TOUCH was not in this list.

closed 2020-04-21 by df8oe

df8oe, 2020-04-21:
I have looked always at wrong python file. It is absolutely logical - if you look into the right place...

PR #1847 (2020-04-21): Added Touch Menu
Fixes #1845