df8oe/UHSDR #1849: mcHF crash when selecting 6m band

mm0gyx, 2020-04-30:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version: 2.11.81
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.5
 * RF Board: mcHF 0.5

### Describe the issue:
A newly built mcHF with all recommended mods, installed bootloader and latest firmware via dfu. Cycling through the bands the mcHF software locks up at the point of selecting 6m band (50 MHz was displayed but radio unresponsive). Radio has to be powered off by pulling the power lead, even after waiting for it to respond for one hour. Tried alternative firmware, most recent firmware that works is 2.11.59 as it doesn't have 6m.

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 SI570 was bought from sdr-kits.com and is a 141dg 160 MHz spec.

mm0gyx, 2020-04-30:
Slawek SP9BSL invited me to open the issue here, apologies if the format is wrong, my first time on github.

mm0gyx, 2020-04-30:
I have another mcHF 0.5 RF/UI, having no trouble with latest firmware. Obviously the SI570 is causing the trouble, is there an option to remove 6m from band map? I couldn't see it.

df8oe, 2020-05-01:
Hi,
there is no possibility to remove it from the map if at boot time detection of SI570 tells "working on 6m". We have not have this before. If SI570 does not work you get red frequency digits and non-working TX/RX capabilities - but no system crash. Very curious. Maybe you can solve it by reducing I2C speed for SI570? Othe possibilty is that you have a scrambled config. To test this try starting with "clean frequency memories" and if that does not work with "complete factroy default config". It is a good idea first to backup your existing configuration via menu so that you can return to it if the issue is persistent in both cases.

db4ple, 2020-05-01:
Hi, I am also wondering what is going on. All our I2C calls have a timeout, meaning, if they have issues to complete, they should return with an error code instead of hanging there indefinitely.
@mm0gyx: I would like to get the full name/markings of the SI570.

mm0gyx, 2020-05-01:
@db4ple Hi, I have the following data from SI570:

Si LABS 570
CAC000141G
D0LOK007+

@df8oe Sorry to say, I'm unsure how to perform factory default config, could you hint where to look?

df8oe, 2020-05-01:
https://github.com/df8oe/UHSDR/wiki/Operating-Manual#key-press-map

mm0gyx, 2020-05-01:
Thankyou, I performed the factory default config, there's no change to the problem with latest f/w. I rolled it back and re-installed saved config.

db4ple, 2020-05-01:
Ok, could you please take a short video clip when you switch from 10m to 6m to see what (not) happens at this point. You can just drag and drop the video file into a comment. THX!

df8oe, 2020-05-01:
Please try to start from 10m and increase the frequency until you left the supported range of your SI570. Best way to do that is to activate "dynamic tune" and set the steps to 100KHz. Is the device locking up? Or just turn the frequency digits red? Or does tuning to 6m frequencies now work?

mm0gyx, 2020-05-01:
Using dynamic tuning I got to 39.360 MHz before it locked up, number remained white.

df8oe, 2020-05-01:
This is related to the SI570. If you exceed its limits it cannot tune to the desired frequency. BUT: It should not hang. I think it is defective or it is an incompatible version. We never have had that with any SI... so I think it is defective.

mm0gyx, 2020-05-01:
Hmmm, maybe...I'll live with it and stick to f/w that works for me, I have no interest in 6m band with mcHF. Thanks for your time.

Ian MM0GYX

db4ple, 2020-05-01:
Hi, as @df8oe wrote, since the SI570 not even reaches its official 40 Mhz (4*LO = 160Mhz) limit, there is surely something defective or not properly done on the hardware side of things. This may also affect the normal operation of the TRX when staying in the "normal" bands. The higher we go with the SI570 frequency, the more power is being used. It may be worth trying to resolder the SI570 just be sure power supply is not causing the issue. 
Is it always the identical frequency at which it crashes? Or does it vary?

mm0gyx, 2020-05-01:
The range of frequencies that it hangs at are within 20 kHz of each other.

I will tear it apart and reflow the SI570 and report findings.

mm0gyx, 2020-05-01:
Gentlemen, when I was a younger man I would have said, "it can't be my soldering, it is perfect!". It was my soldering! Working as advertised, thanks for giving me the push to rework it.

db4ple, 2020-05-01:
Good news! Happy to have pushed you!

df8oe, 2020-05-01:
Please close the issue now.

closed 2020-05-01 by mm0gyx