df8oe/UHSDR #1851: PSK mode

SP5QAD, 2020-05-10:
# UHSDR Firmware Defect Issue (Bug) Template 1.1
## Issue Data goes here (please remove text above):

Your firmware version: 2.10.1
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:

Hello, I'm testing new firmware (v2.11.81). There seems to be a problem with the PSK mode. The change from BPSK31 to BPSK63 is visible (marker spacing) after a change e.g. from PSK-L to PSK-U and vice versa. In addition, it does not decode transmission. RTTY mode works fine.

I returned to version 2.10.1 (theoretically stable). Similar behavior here, but a little better: BPSK transmissions are decoded, but the visual change in marker spacing (after a change from e.g. 63 to 31) is visible only after changing the modulation type and returning again (or a change e.g. when working PSK-L on PSK -U, and return to PSK-L again).  Any suggestions?

--
73, Marcin

df8oe, 2020-05-11:
Please try reducing frequency adjust step to 1Hz and finetune RX frequency very exactly. Because of missing horse power compared to computers there is no AFC or equivalent. You have to tune to frequency absolutely exact.

vy 73
Andreas

SP5QAD, 2020-05-11:
That's right, especially with PSK. (At RTTY, from what I observed, a bit more tolerance). However, there remains the issue (with PSK) of visual change of marker spacing when changing from 31/64. In one of the previous versions it worked great (I don't remember exactly which one, but maybe I can find the right one).
--
73, Marcin

df8oe, 2020-05-11:
Refresh of markers after changing modes is not working as expected.