df8oe/UHSDR #1859: Band Definitions are not saved

DL8EBD, 2020-05-19:
Your firmware version:  D2.11.84
Your bootloader version: 5.0.4

### Hardware 
SParrow with H7 CPU (480MHz)

### Describe the issue:

Band Definitions are not saved to EEPROM.
After a restart the old (default) 
[uhsdr_config.zip](https://github.com/df8oe/UHSDR/files/4652476/uhsdr_config.zip)

Band Definitions appear:
always R1+2+3

uhsdr_config.json in the attachment

thx
Thomas, DL8EBD

db4ple, 2020-05-19:
That is because no one added the respective information to the information to be saved permanently.
Since I was the one who added the code in the first place, maybe it is time to do this now.

DL8EBD, 2020-05-19:
Hello, Danilo,
that is of course a logical explanation :-)
Thank you and stay healthy

2e0xvx, 2020-05-21:
Hi, first time posting on the group, here are the region 1 pand plan frequencys

region 1 band plan

160 meter
1.810 MHz - 2.00 MHz

80 meter
3.500 MHz - 3.800 MHz

60 meter
5.351 MHz - 5.366 MHz

40 meter
7.00 MHz - 7.200 MHz

30 meter
10.100 MHz - 10.150 MHz

20 meter
14.000 MHz - 14.350 MHz

17 meter
18.068 Mhz - 18.168 MHz

15 meter
21.000 MHz - 21.450 MHz

12 meter
24.890 MHz - 24.990 MHz

10 meter
28.000 MHz - 29.700 MHz

2e0xvx, 2020-05-21:
region 2 band plan

160 meter
1.800 MHz - 2.00 MHz

80 meter
3.500 MHz - 4.000 MHz

40 meter
7.00 MHz - 7.300 MHz

30 meter
10.100 MHz - 10.150 MHz

20 meter
14.000 MHz - 14.350 MHz

17 meter
18.068 Mhz - 18.168 MHz

15 meter
21.000 MHz - 21.450 MHz

12 meter
24.890 MHz - 24.990 MHz
here are the region 2 band plan. please correct me if im wrong on any of these, as i'm region 1.
10 meter
28.000 MHz - 29.700 MHz

2e0xvx, 2020-05-21:
now that i have posted all three region bands, would it be possible to put the band edges on the screen, maybe a thin red line under the spectrum part. much like the sdr sharp displays the band scope. or, when you select your region, and you select the "off" on transmitt out of scope, it would block the ppt.
Keep up the good work guys, bloody loive my rig. 
region 2 band plan

160 meter
1.800 MHz - 2.00 MHz

80 meter
3.500 MHz - 4.000 MHz

40 meter
7.00 MHz - 7.300 MHz

30 meter
10.100 MHz - 10.150 MHz

20 meter
14.000 MHz - 14.350 MHz

17 meter
18.068 Mhz - 18.168 MHz

15 meter
21.000 MHz - 21.450 MHz

12 meter
24.890 MHz - 24.990 MHz

10 meter
28.000 MHz - 29.700 MHz

FK8IH, 2020-11-29:
I plan to build a mcHF and use it in Thailand where they allow only transceivers limited to 3.5 to 3.6 MHz and 7.0 to 7.2 MHz for the 80 and 40M bands, the 60M is not allowed, the other bands have the same limits as EU.
I guess the Hambands edges are already written in the firmware as the mcHF manual operator's manual reads page 5 "Band Display: To the right of the main display is an indicator of the amateur band in which the current frequency is tuned. If the current frequency is outside an amateur band it will display “Gen” (e.g. “General Coverage”)."
Can the firmware be modified in order to allow the user to set by menu the band edges of his IARU region or when needed his country as he can do with ICOM IC-705 and IC-7300? ICOM allows the user to 1. mark the band edges, 2. ear or not ear a beep when he reaches an edge, 3. allow or prohibit transmission (PTT signal?) outside of a Hamband. Such a modification to the firmware by the addition of a configuration menu would be nice and would allow the spread of this nice transceiver and its UHSDR firmware in the whole world.
73 - Pierre - FK8IH

oe1eax, 2020-12-22:
Hallo,  first post here from my side, thanks for this excellent project !

Same issue, Band definitions not saved, after reboot R1+2+3 is in place

mchf RF Board V06.3 
rig assembled by O0EMT with  fw 2.11.89
firmware version: 2.11.92 
bootloader version: 5.0.4 

anaother thing, but i think thts not sw related: no swr  readings on display....  

73 - Bert OE1EAX

db4ple, 2021-02-19:
This has been fixed, issue closed

closed 2021-02-19 by db4ple