df8oe/UHSDR #1864: Lpf coils inaccurate  value

Pd9rb, 2020-06-12:
Question:
At this moment I am building my mcHf, bought the diy kit from Chris (M0nka) and just made all torroids for the LPF according to the table.
But when I measure them I get a much higher value than it should be e.g.: 8turns on a T37-2 core gives 0.37 µH instead of 0.26 µH indicated in the schematics, when I remove 2 turns give around 0.28 µH, 27turns gives 3.6 µH  instead of 3 µH that it should be and have to remove 3 turns to get close to 3 µH I have this effect on every coil I made.
Spreading the wire over the core has only a very small effect on the measured value.
This is strange because I expect that the kit is build by so many did they have the same problem?
My lrc meter is a DER EE De-5000 that has always been(and still is) very accurate and never failed me, any ideas what is the best to do? Or made them as in the table or use the schematic value and modify them?
Thanks,
Luuk (Pd9rb)

sp9bsl, 2020-06-12:
Hi Luuk,
remember that the cores have tolerance at about 20%, so, your 3.6uH is within this range. Also there is a lot of core fakes on the market, works ok but have little different permability. With the lower inductance value around hundreds of nH, you will have bigger error because of wires you use for measurement. Make the inductors based on schematics with your inductance meter ratcher , than use the turns from the table. The final inspection of the filter should be using a VNA - for example nanoVNA - good to have one, it's price is 5 times lower than you DER EE...

73 Slawek

sp9bsl, 2020-06-12:
You have the answer at mchf group.io, so please close this issue.

closed 2020-06-12 by Pd9rb