df8oe/UHSDR #1866: mic gain issue

bybra73, 2020-06-22:
Hi,

all FW starting with 2.11.78 and newest have mic gain bug

Mic is not connected
Only PTT button connected to radio

if I set mic gain lower than 51 all OK
If I set mic gain 51 or more I have this bug
I attached link to the video
https://www.youtube.com/watch?v=TGyW8vILLvE

I remember that Mic is no connected all this time

All FW before 2.11.78 work fine without this bug

73! Alexey UT0UM

bybra73, 2020-06-22:
i.e. with mic gain 51 or more you can see some bug like auto oscillation

bybra73, 2020-06-23:
this bug appear only if I set Dynamic mic in menu
if I set Electret mic al OK

db4ple, 2020-06-23:
Thanks for reporting. Without being an expert on this matter hardware-wise, I can tell that above 50 mic gain we enable the WM8731 codecs internal MICBOOST extra gain. 

I would assume that the extra gain leads to some kind of oscillation in your hardware  setup.  I haven't tested it in my setup.  And I don't think this is anything we can deal with in software. Especially since it probably only happens if there is no microphone connected and in this case simply select a different input or keep the gain below 51 solves the problem. Please be so kind and try what happens if a microphone is connected... THX!

df8oe, 2020-07-15:
Is it present even if a microphone is connected? If not: please close this issue - thank you.

closed 2020-07-23 by df8oe