df8oe/UHSDR #1869: what is the meaning "Nothing to build for mchf-eclipse"?

ideapad2009, 2020-07-22:
project can not compile.

df8oe, 2020-07-22:
Please can you specify your workspace? Graphical tool or build via make? I am building via make and it works perfectly.

ideapad2009, 2020-07-22:
I import the archive into eclipse IDE and use the Build  menu. It give the tip  "Nothing to build for mchf-eclipse" in console.

df8oe, 2020-07-22:
I haven't build using IDE since years - only building via make. Have you build mchf-libs first? Every build consists of two steps. First the libraries, then the firmware...

ideapad2009, 2020-07-22:
mchf-libs also can not build. The ArtifactType of "Build ArtifactType" menu is null instead of "Executable".

db4ple, 2020-07-22:
Hi, let me ask you a couple of questions, please answer ALL of them:

Did you follow the instructions given in the UHSDR Wiki to install GNU MCU Eclipse etc.? 

If not, please do so and try again with this Eclipse installation. Is it working now?

If you did install GNU MCU Eclipse and it is not working, please paste the screen shot of your Eclipse project view and the "Help->About Eclipse" window. 

THX!

ideapad2009, 2020-07-23:
> Hi, let me ask you a couple of questions, please answer ALL of them:
>
> Did you follow the instructions given in the UHSDR Wiki to install GNU MCU Eclipse etc.?
>
> If not, please do so and try again [...]

Hi,I do exactly as the guide again.Things changed, Build Error information: Program "arm-none-eabi-gcc" not found in PATH. I am sure I have installed arm-none-eabi-gcc.

db4ple, 2020-07-23:
Hi, verify that on Window->Preferences->MCU the "Global ARM Toolchain Paths",  "Global Build Tool Paths", "Global OpenOCD Path" point to the right directories/settings where the respective tools are installed.

[image]

ideapad2009, 2020-07-23:
> Hi, verify that on Window->Preferences->MCU the "Global ARM Toolchain Paths", "Global Build Tool Paths", "Global OpenOCD Path" point to the right directories/settings where the respective tools are installed. Paths will not be indentical to the picture shown below, but you get an idea. The paths usually end with ".../bin"
> 
> [image]

Per your tip,I have sucessfully compiled. Thank you.

closed 2020-07-23 by ideapad2009