df8oe/UHSDR #1871: UiDriver_StartUpScreenInit() passed argument but function does not seem to take an argument

rogerclarkmelbourne, 2020-08-02:
UiDriver_StartUpScreenInit(2000);

https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/src/uhsdr_main.c#L363

https://github.com/df8oe/UHSDR/search?q=UiDriver_StartUpScreenInit&unscoped_q=UiDriver_StartUpScreenInit

I suggest that you change all prototypes for functions which don't have any arguments, to accept type void, as the compiler will then give an error when you pass an argument to a function which does not take one.

PS.
In this case, I presume that the UiDriver_StartUpScreenInit(2000) was probably intended to hold for 2000ms but it doesn't do this as there isn't a call to delay in UiDriver_StartUpScreenInit

db4ple, 2020-08-02:
Roger, thanks for making me aware of this difference between C and C++ (in C++ ` int f()` and `int f(void)` have the same interpretation, but obviously not in plain C. 
So you are right, we should add void in the appropriate definitions.

db4ple, 2020-08-02:
BTW, read more about this here: https://stackoverflow.com/questions/693788/is-it-better-to-use-c-void-arguments-void-foovoid-or-not-void-foo

rogerclarkmelbourne, 2020-08-02:
Hi Andreas

I was not aware of this problem on the OpenGD77 until Daniel F1RMB told me, sent a PR for the change.

Luckily I think in the OpenGD77 firmware we didn't have any places where args were passed to a void function

df8oe, 2020-08-02:
Hi Roger,

this is a very important issue you have worked out and it is not easy to debug - because you do not see an issue if you look at the code. Many thanks for this contribution!

rogerclarkmelbourne, 2020-08-02:
I'm glad to be of some small help.

db4ple, 2021-02-19:
We mostly did fix this issue by adding void to many function signatures.

closed 2021-02-19 by db4ple