df8oe/UHSDR #1873: HY28B parallel mode - UI board 0.6.3

Radek-2, 2020-08-29:
mcHF - PCB rev. 0.6.3
firmware - UHSDR 2.11.89
bootloader - UHSDR  5.0.4
CPU - STM32F437VIT7
LCD - HY28B

Hello, I'm trying to connect the HY28B display in parallel 16bit mode to PCB revision 0.6.3, which has only the SPI interface by default. On the PCB I made wire jumpers between the CPU and the LCD socket according to the rev. 0.6.0, I connected the LCD to parallel mode by setting jumpers IM0 ... IM2, and I disconnected the SPI SDO and SDI lines.
After turning on the screen **remains stuck on the splash screen**, the transceiver in addition seems to work normally. Bootloader screen is also working OK.
So far I have tested several different versions of the firmware from archive, another HY28B display, I checked the connection twice, tested without RF board. Previously, the display was connected via the SPI interface and worked well. The reason why I want to switch to a parallel interface is the connection of a 3.5 "LCD ILI9486, which has a better refresh rate via the parallel interface than via SPI.

[image]
[image]

Any advice?
Thanks
vy 73 Radek

db4ple, 2020-08-29:
Around this point you are showing, the touchscreen gets activated. Bootloader and splash screen don't use the touchscreen and thus are not initializing the controller. You may want to double check the touchscreen SPI wiring.

Radek-2, 2020-08-29:
Touchscreen SPI wiring checked. Compared 0.6.0 to version 0.6.3, there is a change - the possibility of connecting CS and IRQ to PA14, PA13 instead of PA4, PA9. I tried to connect these lines but still without change :( The remaining lines of the SPI interface are unchanged so the current connection is as follows:

CS - LCD pin 32 - U4 PA9 pin 68
IRQ - LCD pin 36 - U4 PA4 pin 29
SCK - LCD pin 33 - U4 PB13 pin 52
SDO - LCD pin 35 - U4 PC2 pin 17
SDI - LCD pin 34 - U4 PC3 pin 18

I have the firmware compilation tools (Eclipse IDE) installed - is there an option to turn off touchscreen support ?

[image]

Radek-2, 2020-08-29:
To **check** the SPi connection of the **touch screen**, I disconnected the RD pin, switched the display to **SPI mode** and the display **including the touch screen** works normally. In parallel mode still stuck on the splash screen.
And as far as I know, the connection of the SPI interface for the touch screen remains the same for both parallel and SPI LCD mode.
[image]

df8oe, 2020-08-29:
I do not think it is a touchscreen wiring problem. Many HY28B which are produced after ~2017 have big problems working in parallel. That is the reason your PCB does not have layout for parallel use. Can you check at another mcHF which has a working parallel LCD if your LCD works in this mcHF and the other LCD works/stucks at yours?

Radek-2, 2020-08-29:
Unfortunately, I do not have the opportunity to try the display on another transceiver. I didn't even know about the problems with the HY28B parallel interface....
I have purchased a 3.5 "44 pin display with ILI9486 so I will probably buy an LCD 3.5" adapter board for the OVI40/MCHF and I will hope it will work with a parallel interface ....

df8oe, 2020-08-29:
The "parallel problems" are only known on HY28B...

db4ple, 2020-08-29:
@df8oe : I don't think your theory about the data traffic applies here, as the display stops working before the final "text" ("starting up normally") is displayed. Until then there is "not much" traffic. But it might still be a "parallel mode issue".

@Radek-2 : If you can successfully compile and flash firmware, I suggest to try to identify the place when the code stops executing when using parallel mode.
Easiest way is to move the LED off code in src/uhsdr_main.c from line 441 to earlier places:
https://github.com/df8oe/UHSDR/blob/9e5c031445b15750948a207b32bf184715fb9dda/mchf-eclipse/src/uhsdr_main.c#L441

BTW, how do the leds appear when stuck in parallel mode?

Radek-2, 2020-08-29:
@db4ple : When the splash screen appears, both LEDs are turn ON up for a few seconds, then the red one turns off - at that moment the display flashes once, the splash screen remains on the display and only the green LED is on.
The firmware of the transceiver probably continues to run, if I connect headphones, I can tune the SSB broadcast on the band, regulate the AGC, display backlight, etc ...
And yes, I can successfully compile the firmware.

df8oe, 2020-08-30:
I forgot the sentence "...starting up normally..." - sorry. But my logical conclusion that firmware moves over to "normal operation" and does not stuck still alives. For me it is looking as a timing problem (hardware - LCD controller on LCD panel). The reason "data traffic much higher" was definitely wrong - but my thoughts "HY28B does not work properly in parallel mode" still persists.

db4ple, 2020-08-30:
@Radek-2 : Ok, based on your observation that the radio essentially boots into normal operation with a stuck splash screen display I am now taking side of @df8oe: it seems to be an interesting issue related to the parallel mode operation of your particular display. 
I would assume that changing the display to a different type should solve the problem as the basic parallel mode connection does not seem to be an issue (otherwise bootloader and splash screen would not show up).

Radek-2, 2020-09-07:
Hello,
today I finished a simple single-sided PCB for a 44-pin 3.5 "LCD adapter with ILI9486 with a parallel interface (the board also allows you to connect Rpi SPI LCD).
However, the situation with the parallel interface is still the same, the LCD works in bootloader mode, but when the TRX is started, the image stuck on the splash screen :(
What is your opinion on replacing the microcontroller?
[image]
[image]

df8oe, 2020-09-08:
I do not think this is an MCU related issue. All data lines are working, Maybe a soldering issue (one/some line has resistance >> 0R) - but that would be very strange. I never have had it. But much mote it is a soldering issue than the MCU itself.

sp9bsl, 2020-09-08:
Hi,
looks like there is issue with something which overwrites the lcd data interface, for example the spi interface for touchscreen, briefly looking at the schematics of v0.6.3 i see the signal LCD13 (PB13: SPI SCK) is directly connected to WR PIN of LCD, you must check if you have disconnected it. Loking at the code - after displaying startup screen, there is TS init and some more peripherials init like encoders. It might be that you have short circuit to other lcd pins.

As colegues wrote we have plenty of lcds working this way in mcHF and OVI40 UI (since 2017) and noone rised such display issue, so please again: recheck all the lcd connections against short circuits...

sp9bsl, 2020-10-19:
@Radek-2 being just curious - did you solved the issue?

db4ple, 2021-02-19:
@Radek-2: If you still need help, please reopen this issue. We're closing it due to inactivity for now.

closed 2021-02-19 by db4ple