df8oe/UHSDR #1881: CW LED settings not saved

WD8BXS, 2020-11-27:
Your firmware version: **2.11.91**
Your bootloader version: **5.0.4**
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x   **6.3**
 * RF Board: mcHF 0.x / OVI40 1.x  **6.3**

### Describe the issue:
**I just noticed that the CW LED on/off settings will not save.
Once turned off in the CW menu and unit is powered off then on, the LED is back to the on setting.

I wonder if this can be fixed??

Thanks, vy 73,
 de WD8BXS, Chuck**

db4ple, 2020-11-30:
I worked on this, needs some testing. With that change all relevant CW decoder settings (those which can be controlled via menu) will be saved.

WD8BXS, 2020-12-01:
Oh WOW, thanks so much, I will keep an eye out for the firmware update!

 

De Chuck

closed 2020-12-04 by WD8BXS