df8oe/UHSDR #1886: PA bias is 0, TX not possible

rpmeier, 2020-12-24:
Running UHSDR Vers. D2.11.91.

Early this week received mcHF and first two days running smoothly. Today mcHF did no start. Unplugged powersupply and tried again. No luck. Did a hard reset with F1,F3,F5. 
Now receiving the following message during start up:
"PA bias is ), TX not possible. Boot delay because of errors or warnings".

Any suggestion how to solve above?

df8oe, 2020-12-24:
You have to set up configuration completely from the scratch.

This affects PA bias setings, power settings, IQ-settings and so on. F1-F3-F5 resets them all to defaults (ans PA bias to ZERO due to decurity...)

vy 73
Andreas

rpmeier, 2020-12-24:
Thks, before starting do you recommend to upgrade to latest sw version, 2.11.94?

df8oe, 2020-12-24:
Not mandatory, but of course you can do it. It will not impact your settings.

But it is a wise idea to backup your configuration to STM flasharea using menu function. So you can restore your settings if you do have a saving issue once again. Mostly this is caused by power failing during power-off-process (saving data to EEPROM...)

vy 73
Andreas

rpmeier, 2020-12-24:
Am I correct that I directly can do firmware upgrade via usb? I have downloaded the bin file and put it on usb. The usb drive is detected with band - and power. But I get an error message: error code 4 problem writing fw-mchf.old. Any suggestion? Sorry for asking but probably need steep learning curve :-).

rpmeier, 2020-12-24:
> You have to set up configuration completely from the scratch.
> 
> This affects PA bias setings, power settings, IQ-settings and so on. F1-F3-F5 resets them all to defaults (ans PA bias to ZERO due to decurity...)
> 
> vy 73
> Andreas

Andreas, it is not possible to receive a "factory"bin file as it came out of the box?

df8oe, 2020-12-25:
The "factory bin" is what you get when you start with F1+F3+F5. All settings are individual and you *must* walk through the configuration steps now. The only thing what could have been prevented that is "you have made a configuration backup" via menu in your STM32 flash area.

It is like a hdd break at your PC or laptop: you are lucky if you have made a backup...

Flash size of mcHF is 512KB or 1024MB. Recent firmware fits into 512KB MCUs.

Best USB sticks for mcHF upgrades are very old ones - like the 1GB you used. Take always this in the future.

rpmeier, 2020-12-25:
Moin Andreas, clear. 
Fröhliche Weihnachten und einen guten Rutsch ins Neue Jahr.

df8oe, 2020-12-25:
I wish you the same - stay healthy!

vy 73
Andreas

closed 2021-01-31 by df8oe

Nandell, 2024-01-27:
Hello anybody. I`m totally new in the SDR-Szene. My TRX comes from china. It works fine the first 3 weeks. Last weekend i had a qso in cw, while the TRX started new and show me the following error:
"PA Bias is 0, TX not possible. Boot Delay because of Error or Warnings" 
My internet research gives not successfully solutions. Because my know how about this SDR-Technic is`nt yet enough. Where i find a step by step instruction with a easy way to repair the littel box ? The Software-Reset with F1/F3/F5 was`nt helpful. PA Bias is 0 and not possible to chage it.
If somewhere is a german language groupe, please give me a link too.
Thank you for the help.

df8oe, 2024-01-27:
First look at the WIKI:
https://github.com/df8oe/UHSDR/wiki

If you have not found a solution, here is a German discussion group::
https://www.amateurfunk-sulingen.de/forum/