df8oe/UHSDR #1887: Update 2.11.93, 2.11.94 does not work

g7jiq, 2020-12-25:
Your firmware version: 2.11.92 (working)
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
First release of mcHF with 512kb cpu, 
 * UI Board: mcHF 0.x unknown
 * RF Board: mcHF 0.x unknown

### Describe the issue:

Has been working ok with all earler updates,
Updates 2.11.93, 2.11.94 stick on boot screen with flashing red/green leds,
reloading 2.11.92 works fine.

df8oe, 2020-12-25:
very interesting.... This needs confirmation. I do not have access to 512KB MCU since a couple of years and cannot check...

db4ple, 2020-12-25:
Looking at the changes we did between 2.11.92 and 2.11.93 I doubt it is a 512k issue.

@g7jiq : Did you ever use an "unofficial" UHSDR firmware on this machine (like the ones posted in some forums with changes?). Don't get me wrong, this is absolutely fine.
But as configuration values may have been added there without coordination with the main firmware, this can break things as there may be values left in the the configuration memory which do not work with the newer UHSDR firmware using this data for some different stuff.  Given the fact that we just added one configuration parameter in 2.11.93 this is a plausible explanation. This parameter if set to a wrong value will cause a crash of the machine.

df8oe, 2020-12-25:
@g7jiq : you can check if this is the reason by starting mcHF while buttons F1 + F3 + F5 are pressed. If the radion starts now **it is a configuration scrambling problem**.

db4ple, 2020-12-25:
@df8oe: Good tip. 
@g7jiq : If you pressed F1+F3+F5 => DO NOT TURN OFF MACHINE using Power button, this will overwrite all of your configuration values including calibration etc. with defaults.

df8oe, 2020-12-25:
I just got a feedback that a 512KB machine works with 93 and 94 firmware. Reason for malfunction is not MCU flash size.

g7jiq, 2020-12-25:
Hi, Just tried the F1+F3+F5 and it did nothing, (the LEDs just flashed a different sequence)
Re-flashed it back to 2.11.92 and it works again. (if this is down to the old hardware/mods, I am happy to leave as ver 2.11.92 and only update my newer one)
The only software updates I have used are from your github since ver 2.5.10 (I have them all backed up and this one seems to be the earlyest one I have)

db4ple, 2020-12-25:
@g7jiq:  No need to stick on 2.11.92. We will identify the root cause for the problem and fix it. You maybe not the only one with this issue. With your help we will get this done quickly.
LEDs flashing indicate a crash and basically tell us what happened. Could you record this on video and upload it here (just drag the video into the comment)?

g7jiq, 2020-12-26:
Just to let you you know I have just tried the latest 2.11.95, and it works fine on both my 512 and 1024 cpu radios,
Many thanks to all.

closed 2020-12-26 by g7jiq