df8oe/UHSDR #1904: mcHF Band Edges according to the IARU region and country

FK8IH, 2021-02-25:
A few months ago I raised a question in mcHF@groups.io about the setting of TX Band Edges in a mcHF to comply with the regulations of a country (namely Thailand).
My question was solved end of last December and I found it solved in the last firmware "UHSDR-active-devel". Now I wonder how I could set these THAI band edges in a mcHF. To be more precise there are in this source code the three IARU regions 1, 2 and 3 and added to them the 3 region with Thai specifics, how can I chose the IARU region I want to implement in the setting-up of the mcHF, with what Menu... ?
See : Added Region 3 Thailand and the concept of RX Only Ham Bands by db4ple · Pull Request #1885 · df8oe/UHSDR · GitHub

I just found in UHSDR/ui_menu_structure_mdtable.md at active-devel · df8oe/UHSDR · GitHub 
the following command :

Band Definition (Menu_Conf)
LABEL (NR)                                DESCRIPTION
CONFIG_BANDEF_SELECT     Select which band definition to use for ham bands (Original UHSDR or IARU Region 1 - 3

Where shall I find this command in a mcHF and how can I actually set-up the IARU Region in the mcHF I plan to build?

73 - Pierre - FK8IH

db4ple, 2021-02-25:
Load the newest firmware. Go to configuration menu, find entry "Band Definition", select "Region 3 Thailand". 
Done. Transceiver will have bands according to Thai regulations.

FK8IH, 2021-02-26:
Thank you very much for your answer, which I copied to the mcHF@groups.io
73 - Pierre - FK8IH

closed 2021-02-26 by FK8IH