df8oe/UHSDR #1907: Feature Proposal / Comment Solicitation: Terminal or Remote Mode

bmo, 2021-03-29:
It may be outside the scope of this project for this, but is there any interest in a "terminal" mode for the transceiver where the UI + hardware is used in conjunction with a remote McHF or other transceiver? 

In this way, the radio could be used in a similar way to how K3/0s (out of production) are used today to control other remote radios. 

The McHF could potentially be used with RCForb (http://remotehams.com), and certainly Omnirig in the way described in this post: http://mailman.qth.net/pipermail/elecraft/2021-March/284122.html 

-Brian N9ADG

jimktrains, 2021-06-06:
CAT along with audio or i/q is already supported. I'm not familiar with either of those software packages and dont have an k3. Can you provide anymore details on what enhancements you'd like to see?

bmo, 2021-06-14:
"Remote Mode" would NOT generate any RF locally. It's different from CAT in that it would encode any UI inputs (knobs, buttons, etc.) to the transceiver to emit a stream of characters, which could *potentially* be used to control:

1. Another McHF Transceiver (control/remote relationship)
2. Remotehams.com RCForb software
3. RemoteHamRadio.com client software
4. SmartSDR / PowerSDR client software
5. Any other SDR or amateur radio software that needed "knob" functionality

Today, #2 and #3 can use Elecraft K3/0 (K3/Mini) hardware, which is no longer manufactured. The K3/K3s can also be used in this way -- placing those in "terminal mode" doesn't generate RF, but does generate events via a serial stream, and accepts data for the LCD display and LEDs on the front panel.  

It would be convenient to be able to use McHF in "knob mode" with PowerSDR/SmartSDR software. That software expects a stream of characters in response to the knob direction and velocity.  

Since the McHF can represent an audio input/output device, that would also be available in "terminal mode" to be able to play audio from the remote, and generate an audio stream from the microphone.