df8oe/UHSDR #1908: CW Keyer operation - Enhancement

superDIYer, 2021-03-30:
### Software 
Your firmware version: D2.1.96 (6 Mar 2021)
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 0.6.3
 * RF Board: mcHF 0.6.3

### Describe the issue:
Loving the radio, but as I contemplate taking it out for a POTA activation, I went looking for CW keyer memories and how to activate them and interrupt the tx sequence. Uhhh. Buehler? Anybody? Doesnt exist. 

I'd like to propose a new feature that provides CW keyer memories and operation, potentially to include:
- a new major section of the menu for all the relevant settings
- minimum 4 memories represented in that menu, each capable of taking strings of 100 characters long
- a way to choose transmit mode - single, max # of transmits, unlimited # of transmits
- choose the max # of transmits
- select a memory to activate for tx from a ui key press combination
- implement the ui key press combination to start transmitting
- a means of interrupting the cw memory transmission via either the cw key or the mic PTT, or certain ui board key presses

### Your relevant settings
NA

w1mt, 2021-03-30:
CW memories already exist. Long press M1
https://github.com/df8oe/UHSDR/wiki/Operating-Manual

superDIYer, 2021-03-30:
In fact they do. I wonder if that part of the MAIN operating manual and the quick operating guide could use some updates to point to the RTTY/CW/Digimodes section of the manual?