df8oe/UHSDR #1912: CW straight key mode: use tip contact instead of ring contact

italic-r, 2021-05-16:
Your firmware version: D2.11.81
Your bootloader version: 5.0.4

### The issue:
CW straight key mode requires the plug to be wired to the ring contact instead of the standard tip. Using the ring I cannot connect a mono plug (an old bug, for example) to the radio; it will continually key the radio. Using the tip (or providing a menu choice) will allow use of mono plugs.

### Your relevant settings
CW keyer mode: STR_K
CW paddle reverse: both settings prevent use

db4ple, 2021-05-16:
You may want to use the CW Paddle Reverse, which does exactly that (switch meaning of tip and ring) in the CW Config menu:
 
https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/support/ui/menu/ui_menu_structure_mdtable.md#cw-mode-settings-menu_cw

italic-r, 2021-05-16:
Unfortunately the swap function does not do that for me. The radio still keys when i plug in a mono plug and refuses to key with a stereo wired to tip. If i plug in a paddle and toggle the swap setting, the dits and dahs will swap like i expect. This radio is secondhand so I'm not familiar with the internals yet.

df8oe, 2021-05-16:
That cannot be changed by software. The ring is electrically connected to PTT and so it will key the radio regardless what the software does. If you want to change that you have to open the radio and modify the connetions to the plug.

italic-r, 2021-05-16:
I'll take a look at the schematic this week and see if i can swap that. Thanks.

db4ple, 2021-05-16:
I had a closer look into the software:, Indeed while we switch the meaning of ring and tip regarding the CW side of things, we are not switching the PTT side of things.  If you want to keep the ability to connect a paddle instead of a straight key, hardware modification will be difficult. I suggest to create a simple 3.5mm to 3.5 mm adapter for you mono plug. E.g. from a short stereo 3.5mm extension cable.

df8oe, 2021-05-17:
You can simply swap the tip and the ring track at the paddle plug. This does not change anything in behaviour except 1) you can use a 3.5mm mono plug if you use a straight key (because of the state of the ring is ignored in setting straight key) and 2) you have to swap dit and dah in CW setings for keyer.

italic-r, 2021-05-17:
That sounds like a perfect solution to me, and I'll be looking at doing that over the week. Thanks! I'll close this since it's actually a hardware problem.

closed 2021-05-17 by italic-r