df8oe/UHSDR #1917: Memory Beacon CW mode

ok1if, 2021-08-16:
I have been asking for a long time whether the McHF firmware (RS918) can be set when transmitting CW from the BEACON mode memory. This is a mode where the set time (for example 6sec) is waited after the content of the memory is played and then the content of the message is started to be transmitted again from the beginning. This is important, for example, when giving a CW challenge.

strijar, 2021-11-22:
The firmware version from UB8JDC has a CW beacon mode (https://disk.yandex.ru/d/RLoaIuASHetoLQ)

ok1if, 2021-11-22:
TNx Oleg
Milan

> Od: "Oleg Belousov" ***@***.***>
> Komu: "df8oe/UHSDR" ***@***.***>
> Datum: 22.11.2021 14:20
 
The firmware version from UB8JDC has a CW beacon mode (https://disk.yandex.ru/d/RLoaIuASHetoLQ <https://disk.yandex.ru/d/RLoaIuASHetoLQ>)