df8oe/UHSDR #1918: Band switch only change 48khz (sometimes)

DD8BA, 2021-08-29:
Hello,

when I try to switch the Band my device only jumps 48khz up and down. [Video ](https://www.magentacloud.de/lnk/FFyynZpY) 

I made a video where you can see it, when i go to the menu and change the region or restart the device it works again till i use the tune wheel :(

df8oe, 2021-08-30:
This only can be a hardware issue (solder bridge, no contact, mayne noth only temporarily) or a scrumbled configuration (try to start with a factory configuration set (F1-F3-F5 pressed while switching the radion on).

db4ple, 2021-08-30:
Please let us know which firmware version you are running. And was this firmware updated by you or is it the original firmware?

It is indeed a strange problem. Especially given the fact that changing the region does not really do anything to the data (at least it is not supposed to do so). Normally it should just update the displayed band information if the current frequency changes outside or inside a ham band status in different regions. There is also no code which I can think off, which does add 48khz to an stored frequency. Anyways, let us find out what software you are running.

DD8BA, 2021-08-30:
In the Video i upload you see the FW Version .. its last stable Version 2.12.0

I will try to reset all to defaults like Andreas wrote and report later what happens

kind regards

DD8BA, 2021-08-30:
ps firmware is original and was flashed by Chris M0NKA

DD8BA, 2021-08-30:
In the Video i upload you see the FW Version .. its last stable Version 2.12.0I will try to reset all to defaults like Andreas wrote and report later what happenskind regardsVon meinem/meiner Galaxy gesendet-------- Ursprüngliche Nachricht --------Von: db4ple ***@***.***> Datum: 30.08.21  09:23  (GMT+02:00) An: df8oe/UHSDR ***@***.***> Cc: Steven Sporkmann ***@***.***>, Author ***@***.***> Betreff: Re: [df8oe/UHSDR] Band switch only change 48khz (sometimes) (#1918) 

Please let us know which firmware version you are running. And was this firmware updated by you or is it the original firmware?
It is indeed a strange problem. Especially given the fact that changing the region does not really do anything to the data (at least it is not supposed to do so). Normally it should just update the displayed band information if the current frequency changes
 outside or inside a ham band status in different regions. There is also no code which I can think off, which does add 48khz to an stored frequency. Anyways, let us find out what software you are running.
—
view it on GitHub, or 
unsubscribe.
iOS or 
Android.

db4ple, 2021-08-30:
Please flash the official firmware from here https://www.amateurfunk-sulingen.de/uhsdr/firmware-latest/mcHF/
Instructions are in the Wiki. If you are using the USB pen drive method, you will get a backup of the original firmware, hence this is the recommended approach. Please note that some usb pen drives do not work (these will not be detected). Use a different one in this case.

DD8BA, 2021-08-30:
I use the latest firmware 🙄
i found the problem.. there is a different between a dhort and long press of Band buttons ... short press switch the Band... long press 48khz 🤦‍♂️ but got this after i reset to default setting... and i missing my test equipment here far from Germany

closed 2021-08-30 by DD8BA