df8oe/UHSDR #1925: problem stm32F429 in mode DFU.

AngelSerra, 2022-05-17:
I have installed the stm32F429 by mistake, I have installed the P6 jumper and it only enters the BOOT LOADER mode by pressing the BAND+ button but I cannot get it to enter the DFU mode.  I have done this.

How to enter DFU mode:
1. Connect USB cable to pc (mini usb connector).
2. Close P6 jumper at the bottom of the UI board.
3. Press and hold BAND+ button.
4. Press and hold power.
5. Release BAND+ and keep holding the power button(during the whole update process).
6. USB DFU device should appear.
7. 
but I can't get it to go into DFU mode.
73,
EB5HDT
Help

df8oe, 2022-05-22:
If P6 is installed and connection to STM is correct it is impossible to boot into bootloader mode. Please check connection from P6 to STM.

AngelSerra, 2022-05-25:
I have sent him the board to change it for a stm32f407 the stm32F429 I do not have to desolder, the STM32F429 I have not been able to make it work, included with the original windows 10 STMicroelectronics drivers.