df8oe/UHSDR #1946: Generating binary BIN file size issues

www97221, 2023-09-25:
Thank you for the code file provided by the author. I have a question about how to compress binary files generated by eclipse when using a firmware version that exceeds 600K bytes and only has a firmware size of around 300K bytes

db4ple, 2023-09-25:
Hi, please use the "Issue" function for issues (bugs, improvement suggestions etc.), not for general questions.

To close this issue with an answer: There is never firmware compression at play with UHSDR. 

Due to slightly different build processes firmware generated by a makefile build is generally smaller than Eclipse. However, all builds fit into 1MB of flash, this should not be a problem for most users as the recommended STM32 controllers do have 1MB of flash. 
If you want to know more about this, head over to the forum at https://www.amateurfunk-sulingen.de/forum/index.php?board=19

73!

closed 2023-09-25 by db4ple

www97221, 2023-09-26:
How can I compile bin files into 374K bytes with the same firmware as the release version when the device MCU is 512K ROM

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/26/2023 03:11 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [df8oe/UHSDR] Generating binary BIN file size issues (Issue #1946) |

Closed #1946 as completed.