df8oe/UHSDR #1951: Firmware 2.12.2 for mcHF 0.5 with ESP8266 module

Dirk275, 2023-12-13:
I have a mcHF 0.5 and want to add the ESP8266 Wifi module on the UI board.

Does the latest firmware 2.12.2 recognize the ESP8266 module or do I have to use a specific version of this last firmware for this purpose?
Any chance to get binaries of that version??

df8oe, 2023-12-13:
Sorry - at the moment no support for this module. Additinally there is at the moment no active development. But contributors are always welcome ;)

Dirk275, 2023-12-17:
Thank you for the info, Andreas!
In 2016..18 there must have been a compiler option for using the ESP8266 module. On the schematic of the mcHF 0.5 Chris noted, that after adding the ESP8266 module at its place on the UI board only the firmware would have to be recompiled in order to use the module.
So I thought, that perhaps this compiler-option still resides in the actual firmware.

df8oe, 2023-12-17:
Maybe Chris' version supported this. UHSDR never did.

Dirk275, 2023-12-17:
Ok, thank you.

closed 2023-12-17 by Dirk275